fossasia/pslab-desktop #88: Add all relevant information in /docs folder

mariobehling, 2017-04-07:
The blog is a good place to get people attracted. Documentation itself should be kept in a /docs folder in the repository. So people are able to update it.

anu0012, 2017-04-19:
@mariobehling Can I take this issue?

mariobehling, 2017-04-19:
@anu0012 Sure, please take it up. Please consider that this is not just a simple copy/paste issue, but docs should be updated to reflect the latest developments. We can go step by step as well.

anu0012, 2017-04-19:
@mariobehling Can you please tell me what kind of docs we can have?

mariobehling, 2017-05-07:
Please move the info from these posts into the documentation here:

### Blog posts related to PSLab on FOSSASIA blog 
* [Installation of PSLab](http://blog.fossasia.org/pslab-code-repository-and-installation/)
* [Communicating with PSLab](http://blog.fossasia.org/communicating-with-pocket-science-lab-via-usb-and-capturing-and-plotting-sine-waves/)
* [New Tools and Sensors for Fossasia PSLab and ExpEYES](http://blog.fossasia.org/new-tools-and-sensors-fossasia-pslab-and-expeyes/)

closed 2017-05-26 by mariobehling