fossasia/pslab-desktop #196: Not working with PyQt5

feigaoxyz, 2018-05-31:
In `README` dependency section, we read `PyQt5` should work.
But build with `PyQt5` got some issues along the way:

1. in `Makefile`, utilities are hardwired with PyQt4 tool `pyuic4`:
```
ui_%.py: %.ui
	@echo compiling UI : $@
	@pyuic4 $< > $@
```

2. all source UI code imported `PyQt4`

CloudyPadmal, 2018-05-31:
Hi @feigaoxyz . Thank you for the input! Yes, the source code is extensively using `PyQt4` packages and from [this](http://pyqt.sourceforge.net/Docs/PyQt5/pyqt4_differences.html) article it is read that some of those packages (`QtGui` etc) are version dependent. Let me update the `README` to contain this information.

CloudyPadmal, 2018-06-01:
Updated Readme in #193 which addresses this issue of version conflicts

closed 2018-06-01 by CloudyPadmal