fossasia/pslab-desktop #201: Some of the dependencies aren't available on new systems

harsh-2711, 2018-06-11:
**Current behavior**

Dependencies like qtwebkit aren't available on new Linux system and so they obstructs the app to open

**Expected behavior**

There should be some alternate dependencies available

**Would you like to work on the issue ?**

Yes

harsh-2711, 2018-06-11:
@CloudyPadmal  I made a new version of Experiments file with new dependencies and also modified the file which installs the dependencies while first installing the software. So how should I make PR to keep both of our changes and how can we provide user to switch between downloading the app with these two different Experiments file?

CloudyPadmal, 2018-06-11:
Nice! By installing extra dependencies, will it cause any issues in the system? If not we can have a single file with all the dependencies.

CloudyPadmal, 2018-06-12:
Does PyQt4 not have qtwebkit?

harsh-2711, 2018-06-12:
@CloudyPadmal It wasn't available for me when I downloaded PyQt4. I tried to download it from it's source code but then also it wasn't detected. So, I used qtwebkit from PySide

harsh-2711, 2018-06-14:
@CloudyPadmal Should I proceed for a PR by making a new Experiments file? Then you can combine both as per the need in the main software.

CloudyPadmal, 2018-06-15:
@harsh-2711 Yes please. Let's have a look at it and see!

harsh-2711, 2018-06-20:
@CloudyPadmal I made the experiments file but while modifying dependencies, I found that we need to create a virtual environment to download PySide as given in its documentation. So we need user's interaction. And also Qt dependency is needed to download PySide. So should I proceed with these dependencies?

CloudyPadmal, 2018-06-20:
I'm a little confused about what's causing this much of a mess when installing the desktop app. My PC is also 16.04 LTS. Can you just post the dependency files you've been creating? Are you certain that your PC had PyQt4, not PyQt5? Did you run the installer.sh script or manually installed? I thought the version you were trying is 18.04.

harsh-2711, 2018-06-21:
@CloudyPadmal You can close this issue now as when I installed the application again it worked. I don't know what was the problem before. It might be I have been using a different version then.

closed 2018-06-21 by harsh-2711