fossasia/pslab-desktop #203: Upgrade PyQt4 to PyQt5

CloudyPadmal, 2018-06-21:
There are many compatibility issues occuring due to the old version of PyQt libraries. They need to be upgraded to use the latest version to avoid issues as such.

cweitat, 2018-06-26:
Would best if can take this opportunity to upgrade to Python 3 so that we won't have such issues.

CloudyPadmal, 2018-06-27:
Yes @cweitat ! That's a good point. I'll take that also into account! :+1:

cweitat, 2018-07-20:
@CloudyPadmal what's the progress of this? Did you face any problems?

CloudyPadmal, 2018-07-20:
Hello @cweitat ! Yes, there were few issues as the app didn't get installed into python 3 and now I have implemented a method to get it installed into python 3. Then I had the PyQt4 version files not getting overwritten in pslab-desktop-apps repo. I tweaked some files and they seems processing. There are 50+ files need editing. I think I'm in the right track!

CloudyPadmal, 2018-07-21:
Hi @cweitat ! I managed to get all the upgrades from python2 to python3 and PyQt4 to PyQt5. But there are method implementation from PyQt4 which are deprecated in PyQt5. As an example, `setMargins`. To support the new `setContentsMargin`, parameters need modifications and casting. So I'm currently stuck at this as there are quiet a lot of these methods need changes.

cweitat, 2018-07-21:
@yatri1609 since you don't have the device, are you able to help Padmal on this?

yatri1609, 2018-07-23:
Ya sure I'll help Padmal on this. Sorry for late reply.

cweitat, 2018-08-11:
@CloudyPadmal is this finished?

cweitat, 2018-08-12:
@CloudyPadmal able to finish by Thursday?

CloudyPadmal, 2018-08-13:
I don't think I could. It includes redesigning of UI components, new object classes as the existing methods are depreciated with the newer version.

fcartegnie, 2019-01-01:
Temporary patch to check and disable helpView
(untested)

```diff
diff --git a/PSL_Apps/bin/Experiments b/PSL_Apps/bin/Experiments
index 0bb6eb7..6385864 100755
--- a/PSL_Apps/bin/Experiments
+++ b/PSL_Apps/bin/Experiments
@@ -17,7 +17,15 @@ args = parser.parse_args()
 from PSL_Apps.utilitiesClass import utilitiesClass
 
 from PyQt4 import QtCore, QtGui
-from PyQt4 import QtWebKit
+
+# Deprecated webkit binding classes not available on Debian/Ubuntu
+global have_QtWebKit
+try:
+    from PyQt4 import QtWebKit
+    have_QtWebKit = 1
+except ImportError:
+    have_QtWebKit = 0
+    pass
 
 try:
     _fromUtf8 = QtCore.QString.fromUtf8
@@ -163,11 +171,14 @@ class MyMainWindow(QtGui.QMainWindow, single_col_exp.Ui_MainWindow,utilitiesClas
                self.menu_entries=[]
                self.menu_group=None
 
-               self.helpView = QtWebKit.QWebView()
-               self.helpLayout.addWidget(self.helpView)
-               self.helpView.setUrl(QtCore.QUrl("../../docs/MD_HTML/index.html"))
-               #from PSL_Apps.utilityApps.indexGen import autogen
-               #self.helpView.setHtml(autogen(pkg_resources.resource_listdir('psl_res','HTML')))
+                if have_QtWebKit:
+                    self.helpView = QtWebKit.QWebView()
+                    self.helpLayout.addWidget(self.helpView)
+                    self.helpView.setUrl(QtCore.QUrl("../../docs/MD_HTML/index.html"))
+                    #from PSL_Apps.utilityApps.indexGen import autogen
+                    #self.helpView.setHtml(autogen(pkg_resources.resource_listdir('psl_res','HTML')))
+                else:
+                    self.helpView = None
 
                self.shortlist=[]
                self.timer = QtCore.QTimer()
@@ -290,10 +301,11 @@ class MyMainWindow(QtGui.QMainWindow, single_col_exp.Ui_MainWindow,utilitiesClas
                        self.col=0;self.row+=1
 
        def setHelpUrl(self,fname): #fname must reside in psl_res.HTML
-               helpurl = pkg_resources.resource_filename('psl_res.HTML',fname)
-               if not os.path.isfile(helpurl): helpurl=None
-               self.helpView.setUrl(QtCore.QUrl(helpurl))          
-               self.tabWidget.setCurrentIndex(1)
+                if not self.helpView is None:
+                    helpurl = pkg_resources.resource_filename('psl_res.HTML',fname)
+                    if not os.path.isfile(helpurl): helpurl=None
+                    self.helpView.setUrl(QtCore.QUrl(helpurl))
+                    self.tabWidget.setCurrentIndex(1)
                
 
        def clearExperimentIcons(self):
@@ -323,8 +335,9 @@ class MyMainWindow(QtGui.QMainWindow, single_col_exp.Ui_MainWindow,utilitiesClas
 
                        self.runningApp = inst
                        #Load help HTML if found
-                       self.helpView.setUrl(QtCore.QUrl(helpurl))          
-                       self.tabWidget.setCurrentIndex(1)
+                        if not self.helpView is None:
+                           self.helpView.setUrl(QtCore.QUrl(helpurl))
+                           self.tabWidget.setCurrentIndex(1)
 
                else:
                        QtGui.QMessageBox.about(self,'Error','Hardware not detected')

```

CloudyPadmal, 2019-05-17:
Fixed in #282

closed 2019-05-17 by CloudyPadmal