fossasia/pslab-desktop #204: Naming Issues

woshikie, 2018-07-06:
[image]

SQR1/2/3/4 Should be renamed into SQ1/2/3/4. There is no SQR labels on the PSLab and it will confuse users.

Also, please use decimals (".") instead of commas (",") for Phase and Duty Cycle. It is confusing because commas is used for commonly used for counting money (each commas after every 3 digits) while on the PSLab,  in this case, we are handling Duty cycle and Phase of Square Waves, which is an arbitrary unit. Therefore decimals should be used.

parammittal16, 2018-12-10:
@woshikie can i do this.

cweitat, 2019-01-07:
dots(".") and commas(",") depends on the region. Some countries in Europe use commas as the decimal separator. For international use, localisation needs to be done.

mariobehling, 2019-08-03:
Closing as we redesigned the UI.

closed 2019-08-03 by mariobehling