fossasia/pslab-desktop #224: PyQt Webkit

TripackMcLovin, 2019-02-11:
Finally managed to get it build on (RHEL) Fedora 27 by remove the distro-type trash from the Makefile and after having nearly all respective libs correct, I ran into:

from PyQt5 import QtWebKitWidgets
ImportError: cannot import name QtWebKitWidgets

but a quick search shows, that QtWebKit is dead, replaced by WebEngine and as long a pslab is stated "in dev" this should be corrected:
https://doc.qt.io/qt-5/qtwebenginewidgets-qtwebkitportingguide.html

If you are willing to provide a clean and proper line-protocol, I could transfer everything into clean c++/qt project and offer the full capability of the board via a qt-lib-project with a proper Example

btw: even when it suppose to use python3, there are some pip-libs runnin in Background based on 2.7.

mariobehling, 2019-02-11:
This is great to hear, but when you test the app you will see that it needs a lot of improvement too. Our goal is to resolve a lot of the issues in GSoC this year. Of course it would be great, if you could help as well.

CloudyPadmal, 2019-05-17:
Fixed in #282

closed 2019-05-17 by CloudyPadmal