fossasia/pslab-desktop #227: Release V1 is not working at ubuntu 18.04 LTS

gdampf, 2019-03-22:
This is still a nightmare! This is not a release - it is garbage!
I set it up with pip install - worked fine so far, but trying to start Experiments leads to this (attached list of packages installed):
`>Experiments
Traceback (most recent call last):
  File "/usr/local/bin/Experiments", line 17, in <module>
    from PSL_Apps.utilitiesClass import utilitiesClass
  File "/usr/local/lib/python2.7/dist-packages/PSL_Apps/utilitiesClass.py", line
 14, in <module>
    from PSL_Apps.templates.widgets import ui_button as button
ImportError: cannot import name ui_button

>pip list
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Ple
ase upgrade your Python as Python 2.7 won't be maintained after that date. A fut
ure version of pip will drop support for Python 2.7.
Package                            Version
---------------------------------- -------
asn1crypto                         0.24.0 
backports-abc                      0.5    
backports.functools-lru-cache      1.4    
backports.shutil-get-terminal-size 1.0.0  
chardet                            3.0.4  
cryptography                       2.1.4  
cycler                             0.10.0 
decorator                          4.3.2  
enum34                             1.1.6  
futures                            3.2.0  
idna                               2.6    
ipaddress                          1.0.17 
ipykernel                          4.10.0 
ipython                            5.8.0  
ipython-genutils                   0.2.0  
jupyter-client                     5.2.4  
jupyter-core                       4.4.0  
keyring                            10.6.0 
keyrings.alt                       3.0    
matplotlib                         2.1.1  
numpy                              1.13.3 
olefile                            0.45.1 
paho-mqtt                          1.3.1  
pathlib2                           2.3.3  
pexpect                            4.6.0  
pickleshare                        0.7.5  
Pillow                             5.1.0  
pip                                19.0.3 
prompt-toolkit                     1.0.15 
PSL                                1.0.0  
PSL-Apps                           1.0.0  
ptyprocess                         0.6.0  
pycrypto                           2.6.1  
Pygments                           2.3.1  
pygobject                          3.26.1 
PyOpenGL                           3.1.0  
pyparsing                          2.2.0  
pyqtgraph                          0.10.0 
pyserial                           3.4    
python-dateutil                    2.8.0  
pytz                               2018.3 
pyxdg                              0.25   
pyzmq                              18.0.1 
qtconsole                          4.4.3  
scandir                            1.10.0 
scipy                              0.19.1 
SecretStorage                      2.3.1  
setuptools                         38.2.3 
simplegeneric                      0.8.1  
singledispatch                     3.4.0.3
six                                1.11.0 
subprocess32                       3.2.7  
tornado                            5.1.1  
traitlets                          4.3.2  
wcwidth                            0.1.7  
wheel                              0.30.0 
`
requirements.txt just content the following:
`pyqtgraph >= 0.10.0
setuptools >= 35.0.2
`

This is meager!

gdampf, 2019-03-22:
Can anybody tell me, what is missing here?

gdampf, 2019-03-22:
pyqt4-dev-tools are also installed (4.12.1+dfsg-2)

CloudyPadmal, 2019-03-22:
Hi, this release is for Ubuntu 16 and below. With the update of PyQt5, some features are not working as some classes from PyQt4 has been removed in PyQt5. We'll make a stable release for Ubuntu 18 with PyQt5 pretty soon.

closed 2019-03-22 by CloudyPadmal

gdampf, 2019-03-23:
What does this has to do with the ubuntu version? - rubbish. But I gave it a try and install the package on Ubuntu 16.04 LTS based on bananapi:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:    Ubuntu
Description:    Ubuntu 16.04.4 LTS
Release:    16.04
Codename:    xenial
```
The result is exactly the same (as expected):
`ImportError: cannot import name ui_button
`
This release is garbage!

techyay, 2019-03-23:
@gdampf Swearing in our issue tracker and insulting contributors will result in you being banned and reported to GitHub on the next incident. If you want to collaborate you can do this in a normal way.

gdampf, 2019-03-24:
Do not threaten me - that is useless!

 

If I tell you, that this release cannot beeing called "release", because ist is buggy like hell and the description is a nightmare, and I call it garbage, it is your personal problem, if you take it as insult!

I am realy upset, because I paid 50 bucks for a brick of hardware - as the software is useless.

 

I do not care about your personal problems. Please report this to Github - I fuck it!

 

----------

You wrote:

@gdampf Swearing in our issue tracker and insulting contributors will result in you being banned and reported to GitHub on the next incident. If you want to collaborate you can do this in a normal way.

 

----------