fossasia/pslab-desktop #228: Release V1 is not working at ubuntu 16.04 LTS

gdampf, 2019-03-23:
This is still a nightmare! This is not a release - it is garbage!
I set it up with pip install - worked fine so far, but trying to start Experiments leads to this:

```
Experiments
Traceback (most recent call last):
  File "/usr/local/bin/Experiments", line 17, in <module>
    from PSL_Apps.utilitiesClass import utilitiesClass
  File "/usr/local/lib/python2.7/dist-packages/PSL_Apps/utilitiesClass.py", line
 14, in <module>
    from PSL_Apps.templates.widgets import ui_button as button
ImportError: cannot import name ui_button

```

This is the same as issue #227 (closed by CloudyPadmal without a solution)!
But this time, I used Ubuntu 16.04 LTS:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:    Ubuntu
Description:    Ubuntu 16.04.4 LTS
Release:    16.04
Codename:    xenial

```

techyay, 2019-03-23:
Insulting developers is unacceptable. The project provides an Android app, that is already very functional. The desktop apps are in prototype stage. We are always looking for people to collaborate with but are not interest receiving insults.

closed 2019-03-23 by techyay

gdampf, 2019-03-23:
insulting? who insults whom? I just listed facts! And the android app is also very buggy (see my comment there). If you are not interested in facts -

Ihr mich auch!