fossasia/pslab-desktop #229: Problems with Concat

YOUR-FATHER, 2019-03-24:
Hello there, 

I tried to run the application, but every time click on "Oscilloscope" to plot the mesurement, this error occure:

Traceback (most recent call last):
  File "/usr/bin/Experiments", line 293, in launchFunc
    inst = FILE.AppWindow(self,I=self.I)
  File "/usr/share/pslab/psl_res/GUI/A_TEST_AND_MEASUREMENT/A_TandM/A_scope.py", line 49, in __init__
    print(self.I.H.version_string+' : ' +params.get('name',"").replace('\n',' '))
TypeError: can't concat str to bytes

I hope someone can help me with this
Thanks!

paltrickontpb, 2019-03-24:
Please try this out:
print(str(self.I.H.version_string)+str(' : ' )+str(params.get('name',"").replace('\n',' ')))

I have just enclosed all the parts to str(). Just put it on the one which is returning byte value. The above code should help, but it isn't good practice. (Just Troubleshooting tip)

YOUR-FATHER, 2019-03-25:
Well, then this problem occure:

Traceback (most recent call last):
  File "/usr/bin/Experiments", line 293, in launchFunc
    inst = FILE.AppWindow(self,I=self.I)
  File "/usr/share/pslab/psl_res/GUI/A_TEST_AND_MEASUREMENT/A_TandM/A_scope.py", line 49, in __init__
    print(str(self.I.H.version_string)+str(' : ' )+str(params.get('name',"").replace('\n',' ')))
TypeError: can't concat str to bytes

I've tried to print each of these substrings seperately,  but i recognize some weird problem; The terminal print this error, equal what is standing in this line 49. Even when it is empty...

AakashMallik, 2019-04-03:
Sadly, this is also the case for logic analyser...

CloudyPadmal, 2019-04-03:
Issue with python3 not supporting the feature https://stackoverflow.com/questions/21916888/cant-concat-bytes-to-str

CloudyPadmal, 2019-05-17:
Fixed with the upgrade to Python3 in #282

closed 2019-05-17 by CloudyPadmal