fossasia/pslab-desktop #231: Error converting installer script compatible with macOS

fragm3, 2019-04-11:
I am trying to convert the installer script compatible with macOS. I am getting this error:
[image]

Any help would be appriciated

fragm3, 2019-04-11:
@CloudyPadmal

orangecms, 2020-06-14:
#568 / `c979fc8f9240fea1b52faaa7d2d43a8b06be3b8a` has added a build script for the desktop app for macOS.

This issue here belongs to https://github.com/fossasia/pslab-python now that the desktop app has been decoupled from it.

Suggested solution:
To install the library system-wide, prefer to use the system's packaging tool. There is a package now available in AUR for Arch Linux, for example. Otherwise, you can keep it in the home directory and use any of the Python tools that set the necessary environment variables or set them directly. See also https://realpython.com/lessons/module-search-path/.

closed 2020-06-14 by orangecms