fossasia/pslab-desktop #232: Electron reinplementaiton of python stack

AakashMallik, 2019-04-16:
## Goals
- [ ] Reimplement the whole Python QT app using Electron.
- [ ] Redesign UI, come up with a unified theme. 
- [ ] Migrate all the features available on android app.
- [ ] Rigorous testing of the newly build up under heavy data load.

fragm3, 2019-04-16:
@AakashMallik sounds like a plan. Let's get started!

fragm3, 2019-04-16:
**Before the GSOC period starts(till end of community bonding period):**

- We plan to create the whole structuring of the dashboard. We plan to use either antd or Material UI for styling our components.
- Create modular and reusable styled-components which can be used across the application
- Create layouts which will help reduce code redundancy and make it easier to manage more number of instruments and components.
- Create oscilloscope component and get feedback from our mentor @CloudyPadmal @mariobehling. Make improvement and use this generalised layout for other instruments.
- Create and implement general dashboard components like help, guide, settings
- We plan to explore electron js more, and make sure the code written is efficient.

**Overall plan:**

Basic Dashboard → Connection to device, electron → Implement Oscilloscope component → Implement other Instruments components →Export Data,  Data visualisation →  User, device settings →  Refactor Code → Documentation

Dev Approach : Implementation, add features → Fix issues, bugs → Finalise the project → Maintenance

closed 2019-08-27 by AakashMallik