fossasia/pslab-desktop #266: Add Flow for Static Type Checking

fragm3, 2019-04-24:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**
Currently, no prop checking is present

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs

* **What is the expected behavior?**
Instead of using Prop Types, add flow for static time checking on the go. As prop type was detached from react, which now needs to installed as a separate module. Flow, developed by React Team(Facebook) is better and more used nowadays.

* **Please tell us about your environment:**
  - OS with version number: macOS

* **Other information** ( Any research that you may have done which you think is the cause of problem )
Link: https://medium.com/@linmic/its-time-to-remove-proptypes-and-just-use-flow-for-react-4a3de615aac5

* **Would you like to work on it?**
Yes

fragm3, 2019-04-24:
@AakashMallik @CloudyPadmal do share your thoughts

AakashMallik, 2019-05-05:
This can really help us in the optimization of the app but to be honest, in the early stages of the app, we are going to face a little bit of problem here and there because of Electron and React stack. We need to iron out those problems first. Get the main components functioning should be our first goal, then we could add this as a secondary goal.

closed 2019-06-10 by AakashMallik