fossasia/pslab-desktop #309: Multimeter Dial component needs simplification and a fix for 0 - 360 issue

AakashMallik, 2019-05-17:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**
- The component, even though functional, needs simplification. 
- The switch case in Multimeter.js should be simplified
- The dial has an issue for 0 == 360 situation. Not a problem, but a little glitch. Needs to be ironed out.
- Code needs refactoring

* **What is the expected behaviour?**
The above-mentioned problems need to be fixed

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
Open for all **beginner friendly**

ojasgulati, 2019-05-18:
@AakashMallik I am working on this 😃

AakashMallik, 2019-06-25:
We have used react-rotary-knob so this issue doesn't exist anymore

closed 2019-06-25 by AakashMallik