fossasia/pslab-desktop #314: Electron process still running after closing the app

ojasgulati, 2019-05-20:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**

     - Electron process after closing still running in the background leaving memory leaks.

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs

[image]

* **What is the expected behavior?**
The app should close without leaving any memory leaks 

* **Please tell us about your environment:**
  - OS with version number: [ Windows / OSX / Linux Distro ]
Windows 10 home x64

* **Other information** ( Any research that you may have done which you think is the cause of problem )

* **Would you like to work on it?**
yes

AakashMallik, 2019-05-23:
@ojasgulati Let add this in the list of discussion on our next meeting. I think it is a good time to address this problem.

ojasgulati, 2019-05-23:
Sure I will try to gather resources to deal this so we can address this problem more efficiently in the meeting.

PR #344 (2019-06-06): [#314] load-balancer version updated 1.0.9, background process cleanup added
Fixes #314 

* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [x] Bug fix
- [ ] Feature implementation
- [ ] Doc updates

* **What changes have you introduced?**
- Load balancer library updated
- Background task cleanup feature added

* **Doe [... cut]
closed 2019-06-06 by CloudyPadmal