fossasia/pslab-desktop #358: Introduce react-redux for global state management for the app

AakashMallik, 2019-06-09:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**
- We are making use of react component state in the whole app as of now

* **What is the expected behaviour?**
- making use of redux will make the app much more developer friendly 

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
Yes

CloudyPadmal, 2019-06-09:
Make sure it is really necessary before proceeding. Infact using redux is kinda proven to make development harder among newbies in industry. Specially in mine, redux is discouraged. Can you list down why we need redux so bad as a comment in this issue? 

https://medium.com/@dan_abramov/you-might-not-need-redux-be46360cf367

AakashMallik, 2019-06-09:
[image]

As in the diagram.. we have a lot of parent level components and state values that are being used deep down in the component tree. For that, we are required to pass them down in a waterfall structure using props. Also for components like Snackbar and Modal, we need to pass the functions to open and close them down the component tree as well. 
There is also a requirement for maintaining a global state value for the device configuration, rather than keeping them separately in different components. This will enable us to update the global state at the starting of the app and this will help us cut the dependency on the device for getting config values every time a component is re-mounted ( that is what we are doing now ). 

As for newbie coders, well... I am trying my best to engage and educate them. There will always be a time to step up from newbie to veteran... ;-p 

@CloudyPadmal

CloudyPadmal, 2019-06-10:
Since our instruments are independent of each other, (most of the time), a global state management is quite an overkill (just for device connection...). Passing props down the component makes it easier to visualize the dataflow.
Anyways it's your call; if it makes development easier..

AakashMallik, 2019-06-10:
> Since our instruments are independent of each other, (most of the time), a global state management is quite an overkill (just for device connection...). Passing props down the component makes it easier to visualize the dataflow.
> Anyways it's your call; if it makes development easier..

That I do agree with. Passing down props always makes the visualization easier. But it is definitely making the codebase a mess, a bit too verbose. So I think I'll go for redux. Hope this doesn't end up being a bad decision.

AakashMallik, 2019-06-12:
## Mandatory
https://redux.js.org/basics/actions
https://redux.js.org/basics/reducers
https://redux.js.org/basics/store
https://medium.com/@tkssharma/understanding-redux-react-in-easiest-way-part-1-81f3209fc0e5
https://redux.js.org/advanced/async-actions

## Side reading
https://redux.js.org/api/bindactioncreators
https://blog.benestudio.co/5-ways-to-connect-redux-actions-3f56af4009c8
https://redux.js.org/recipes/reducing-boilerplate

closed 2019-06-12 by AakashMallik