fossasia/pslab-desktop #393: Power source component freezes if adding a character 'e' in power field

KSNINJA, 2019-06-23:
* **I'm submitting a ...**
  - [x] bug report

* **What is the current status?**
In the power source component, if we try to manually change the power, the field allows us to only add numeric, no characters expect 'e'. On pressing enter, the whole app freezes and then, we have to quit it and start it again.

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs

[image]

* **What is the expected behavior?**
The field should not accept any letter.

* **Please tell us about your environment:**
  - OS with version number:  OSX 10.14.5

* **Would you like to work on it?**
Yes

AakashMallik, 2019-06-23:
@KSNINJA Good job spotting the bug.. check console for error. Also, I think the check function is the culprit here...

KSNINJA, 2019-06-26:
Sorry,  I am a bit busy for a few days.  Please someone else take up this issue.

suyashsingh234, 2019-06-26:
Can i take this issue?

AakashMallik, 2019-06-26:
> Can i take this issue?

Sure @suyashsingh234

commit 62d4bc95 (2019-07-01): fixes #393 (#415)
closed 2019-07-01 by CloudyPadmal