fossasia/pslab-desktop #480: Create the pin layout page as that in Android for the Front layout

AakashMallik, 2019-08-09:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**
- only the backlayout has been done so far

* **What is the expected behavior?**
- The frontlayout should also be covered

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
Yes

rajput2122, 2019-08-22:
I want to work on that issue.can you please allow me to proceed

AakashMallik, 2019-08-22:
> I want to work on that issue.can you please allow me to proceed

Go ahead @rajput2122

rajput2122, 2019-08-22:
Thanks sir

AakashMallik, 2019-08-22:
@rajput2122 Take reference from the back layout component. It should be a copy-paste job from there...

closed 2019-08-28 by AakashMallik