fossasia/pslab-desktop #497: Add empty column lat, long etc to the data logger

AakashMallik, 2019-08-20:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**
- The data that cannot be captured using desktop doesn't have a column in CSV

* **What is the expected behaviour?**
- These data columns should still be provided with empty columns.

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
No

viditkumar, 2019-08-24:
Could you please elaborate the above mentioned enhancement? I am interested in picking this up.

AakashMallik, 2019-08-24:
> Could you please elaborate the above-mentioned enhancement? I am interested in picking this up.

@viditkumar When the user uses the data logger, there are numerous columns that are recorded in a CSV file. When the recording is done in the android app, the latitude and longitude is also recorded, but in the desktop app, we cannot record latitude and longitude. But we still need to put those empty headings from the desktop app.

Shoould be easy enough.

AakashMallik, 2019-08-24:
> @viditkumar When the user uses the data logger, there are numerous columns that are recorded in a CSV file. When the recording is done in the android app, the latitude and longitude is also [...]

PR #500 (2019-08-24): Add blank latitude and longitide columns in generated csv file
* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [x] Bug fix
- [ ] Feature implementation
- [ ] Doc updates

* **What changes have you introduced?**
Fixes #497 
I have added Two empty columns namely longitude and latitude with empty values in them.

* **Do [... cut]
closed 2019-08-24 by AakashMallik