fossasia/pslab-desktop #501: Home screen alignment needs fixing

AakashMallik, 2019-08-24:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**
- The home screen alignment uses centre alignment for an odd number of cards.

* **If the current behaviour is a bug, please provide the steps to reproduce it**
[image]
The robotic arm card is centred and looks weird. 

* **What is the expected behaviour?**
- Left alignment should be used for the odd-numbered card.

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
Yes

priyanshunayan, 2019-08-24:
I want to work on this issue. Can I proceed?

AakashMallik, 2019-08-24:
> I want to work on this issue. Can I proceed?

Yes, by all means. Go ahead.

PR #505 (2019-08-27): Home Screen Cards Alignment
* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [x] Bug fix
- [ ] Feature implementation
- [ ] Doc updates

Fixes #501 
* **What changes have you introduced?**
I have made the cards align to the center in case of even cards in a row but aligned to start if the number is odd. 

* **Does thi [... cut]
closed 2019-08-27 by AakashMallik