fossasia/pslab-desktop #502: Add timebase slider in Logic Analyser

AakashMallik, 2019-08-24:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**
- The user does not have an option to choose between different time bases for logic analyser

* **What is the expected behaviour?**
- A timebase slider should be provided similar to an oscilloscope

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
Yes

priyanshunayan, 2019-08-30:
Can i try this?

AakashMallik, 2019-08-30:
Sure.. go ahead.. @priyanshunayan . This one should actually be really simple.
The value you are aiming to change using the slider is the sleep value in the logic anaalyser python file.

sarthak0906, 2019-10-15:
I would like to work on the issue. Can I try?

PR #547 (2020-06-14): add captureTime slider in logic analyser
* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs hve been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [x] Bug fix
- [ ] Feature implementation
- [ ] Doc updates

* **What changes have you introduced?**
Fixes #502 
I have added the timebase slider in logic analyzer. Earlier it was missing.

* **Does this PR introduce a breaking cha [... cut]
closed 2020-06-14 by orangecms