fossasia/pslab-desktop #503: Change normal graph to brush zoom graph for Logic analyser

AakashMallik, 2019-08-25:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**
- The current graph is a simple one and does not allow the user to zoom in.

[image]

* **What is the expected behaviour?**
- A brush zoom graph can be used instead for added interactive functionality
http://recharts.org/en-US/examples/BrushBarChart
http://recharts.org/en-US/examples/HighlightAndZoomLineChart

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
Open issue for begineers

priyanshunayan, 2019-08-28:
@AakashMallik  I have a doubt in the code. In the following code 
`<Line
                yAxisId="left"
                type="monotone"
                dataKey="voltage"
                data={LA2Data}
                stroke={theme.ch2Color}
                dot={false}
                activeDot={{ r: 4 }}
                name="LA2"
              />`
In the Line Component what is data property doing? In the [documentation](http://recharts.org/en-US/api/Line), I was not able to find data property.

AakashMallik, 2019-08-28:
Tells the Line component, which data object to plot... so this Line component is plotting the values for the second LA2 channel

priyanshunayan, 2019-08-28:
But the documentation tells it doesn't have that property. Is it inherited?

AakashMallik, 2019-08-29:
@priyanshunayan Read this example code. We are dealing with multi-series case.. so we have to follow this example. http://recharts.org/en-US/examples/LineChartHasMultiSeries

sarthak0906, 2019-10-15:
Is someone working on the issue?
If it is still open I would like to contribute to it.

AakashMallik, 2019-10-15:
@sarthak0906 Go ahead...

sarthak0906, 2019-10-17:
Thanks, @AakashMallik 
I started working on the issue but did encounter a bit of problem and needed a bit of assistance from someone.
I was trying to build the app from source but can't make the react to load.
The react dev server starts but the output can neither be seen in the electron app nor the browser at localhost:3000.
Assuming this is caused by the react script as the electron is working just fine.
I have tried running only react-script start and still, the output is the same on the browser.
Also, I have tried both npm and yarn just in case.
About the dev environment, I have tried it on ubuntu system.
I don't have any hardware related to the pslab hardware kit and just want to contribute to the desktop software.

The elements tab in developer menu in electron shows the loaded HTML as:
<html>
    <head></head>
    <body></body>
</html>

M
aking me suspect it is a react problem, although it can be other any error.
Help needed.

sarthak0906, 2019-11-02:
hey @AakashMallik can you please help me with the issue,
still can't figure out

JigarMarvaniya, 2020-02-06:
Is someone working on the issue?
If it is still open, I would like to get it a try.

JigarMarvaniya, 2020-02-10:
Anyone ?

hqrshguptq, 2020-08-20:
Is it fixed @AakashMallik  , else would surely like to donate..
Thanks :D

AakashMallik, 2020-08-21:
@hqrshguptq My write access to the repo has been revoked. I am not a maintainer of this project. Sorry.