fossasia/pslab-desktop #504: Fix type in Issue and PR templates

AakashMallik, 2019-08-25:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**
- There are typos in the issue and PR templates

* **What is the expected behaviour?**
- Typos need to be fixed 

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
Open issue

kushthedude, 2019-08-25:
@AakashMallik Use tags `good first issue` or `help wanted`, As these tags are read by GitHub to the new contributors in the repository and shown under `collected issues`

AakashMallik, 2019-08-25:
@kushthedude Point taken

priyanshunayan, 2019-08-26:
@AakashMallik I am not able to find any typos. I even tested them on grammarly in case I miss any. Can you tell me what's wrong there? I am worried because I have created those files.

AakashMallik, 2019-08-26:
@priyanshunayan 
typo: behavior 
location: issue_template

priyanshunayan, 2019-08-26:
@AakashMallik But I have seen 'behavior' in use at multiple places. I have attached the screenshots.  
[image]
and in google dictionary, in front of "noun":
[image]

kushthedude, 2019-08-26:
@AakashMallik American and British😂

AakashMallik, 2019-08-26:
> @AakashMallik But I have seen 'behavior' in use at multiple places. I have attached the screenshots.
> [image]
> and in google dictionary, in front of "noun":
> [image]
> 
> I think both are used interchangeably.

yup, you are right about that... So should we use American English or British English?

mariobehling, 2019-08-27:
Ok, thanks for the input. Let's focus on code now. I think we are all clear and can fix typos on the way. Thank you.

closed 2019-08-27 by mariobehling