fossasia/pslab-desktop #511: Data writing to CSV for Logic Analyser

AakashMallik, 2019-08-28:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**
- Currently missing

* **What is the expected behaviour?**
- This feature should be added

* **Please tell us about your environment:**
  - OS with version number: Ubuntu 18.04

* **Would you like to work on it?**
Yes

PragatiVerma18, 2019-09-06:
Hi there!!! Can I work on this? And can you please guide a little more about what files do I exactly need to modify to achieve this and what enhancements do you need?

AakashMallik, 2019-09-06:
> Hi there!!! Can I work on this? And can you please guide a little more about what files do I exactly need to modify to achieve this and what enhancements do you need?

This issue has been addressed in an earlier PR.

closed 2019-09-06 by AakashMallik