fossasia/pslab-desktop #512: Make all files in install branch build from dev and master without LFS

mariobehling, 2019-08-29:
Master builds are currently still using LFS. Win images are too big to get build without LFS.

1. Build master images without LFS
2. Investigate, propose and implement ways to reduce the size of the .win image to below 100MB or implement a way to automatically upload the .win image to Dropbox during build using Travis and environment variables.

CloudyPadmal, 2019-08-29:
Something similar https://medium.com/@baymac/continuous-integration-of-latex-documents-using-travis-ci-a1916c89e978

mariobehling, 2019-08-30:
First step is to create a dropbox account. Ones things are working this dropbox account can be assigned to FOSSASIA.

orangecms, 2020-11-01:
no longer applicable - we are using GitHub release attachments now

closed 2020-11-01 by orangecms