fossasia/pslab-desktop #517: Cannot install .deb package from install branch

mariobehling, 2019-08-29:
Deb package from install branch does not install. It gets stuck at 17%.

[image]
``` sudo apt install ./pslab-desktop-development-linux.deb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'pslab' instead of './pslab-desktop-development-linux.deb'
The following packages were automatically installed and are no longer required:
  libllvm7 linux-headers-4.15.0-44 linux-headers-4.15.0-44-generic linux-headers-4.15.0-45 linux-headers-4.15.0-45-generic linux-headers-4.15.0-46
  linux-headers-4.15.0-46-generic linux-headers-4.15.0-47 linux-headers-4.15.0-47-generic linux-headers-4.15.0-48 linux-headers-4.15.0-48-generic
  linux-headers-4.15.0-50 linux-headers-4.15.0-50-generic linux-headers-4.15.0-51 linux-headers-4.15.0-51-generic linux-headers-4.15.0-52
  linux-headers-4.15.0-52-generic linux-headers-4.15.0-54 linux-headers-4.15.0-54-generic linux-image-4.15.0-44-generic linux-image-4.15.0-45-generic
  linux-image-4.15.0-46-generic linux-image-4.15.0-47-generic linux-image-4.15.0-48-generic linux-image-4.15.0-50-generic linux-image-4.15.0-51-generic
  linux-image-4.15.0-52-generic linux-image-4.15.0-54-generic linux-modules-4.15.0-44-generic linux-modules-4.15.0-45-generic linux-modules-4.15.0-46-generic
  linux-modules-4.15.0-47-generic linux-modules-4.15.0-48-generic linux-modules-4.15.0-50-generic linux-modules-4.15.0-51-generic linux-modules-4.15.0-52-generic
  linux-modules-4.15.0-54-generic linux-modules-extra-4.15.0-44-generic linux-modules-extra-4.15.0-45-generic linux-modules-extra-4.15.0-46-generic
  linux-modules-extra-4.15.0-47-generic linux-modules-extra-4.15.0-48-generic linux-modules-extra-4.15.0-50-generic linux-modules-extra-4.15.0-51-generic
  linux-modules-extra-4.15.0-52-generic linux-modules-extra-4.15.0-54-generic
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  pslab
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/103 MB of archives.
After this operation, 575 MB of additional disk space will be used.
Get:1 /home/opntec/Downloads/pslab-desktop-development-linux.deb pslab amd64 0.1.0-806 [103 MB]
Selecting previously unselected package pslab.
(Reading database ... 675685 files and directories currently installed.)
Preparing to unpack .../pslab-desktop-development-linux.deb ...
Unpacking pslab (0.1.0-806) ...
```

CloudyPadmal, 2019-08-29:
Can try `dpkg -i pslab-desktop-development-linux.deb`? :thinking:

mariobehling, 2019-08-29:
Same with dpgk. Nothing still happening after 30 minutes.

[image]

mariobehling, 2019-08-29:
Ok, suddenly works. Yay! But, I am still unclear why it takes so long.

[image]

closed 2019-08-29 by mariobehling

CloudyPadmal, 2019-08-29:
Probably downloading packages in background :thinking: