fossasia/pslab-desktop #529: Power Source UI bugs, negative values, enhance design

mariobehling, 2019-08-29:
There are a number of issues in the UI of the power source.
* using the plus/minus button negative values are possible. Negative values should not be possible
[image]
* Currently there is just a big red line for the value. It is difficult to "navigate" the app. Please help to make it easier by implementing a counter pointer line similar to the Android app.
* Pressing plus or minus should continuously increase or decrease the value. Currently the user has to press again and again to change it.
* Please integrate the possibility to change values into the UI directly instead of having a popup. When the user clicks on the number, the number should be editable and the popup info can be shown as well (e.g. value Volt limits)
[image]
* Saving feature does not work if device is not connected

Dharmesh-Poddar, 2019-09-09:
@mariobehling @AakashMallik can I work on it.

AakashMallik, 2019-09-09:
@Dharmesh-Poddar Sure.. go ahead

Dharmesh-Poddar, 2019-09-11:
@AakashMallik  while making pr I am getting some  issue( React version not specified in eslint-plugin-react settings) 
[image]

AakashMallik, 2019-09-11:
@Dharmesh-Poddar Use 
```
git add .
git commit -m 'msg' --no-verify
git push origin branch_name
```

Dharmesh-Poddar, 2019-09-11:
Thanks for the help @AakashMallik

CloudyPadmal, 2019-09-23:
In power source instrument, negative values are possible with PV1 and PV2 pins.

alcorhythm, 2020-12-20:
Even though it is possible to set negative voltages for PV1 and PV2 using the sliders, the interface doesn't accept negative values in the popup textbox. Cannot enter a minus symbol.

> Pressing plus or minus should continuously increase or decrease the value. Currently the user has to press again and again to change it.

I agree with that. Or is it possible to have a way to change the increment/decrement step used by the + and - buttons. For example a choice between 0.01, 0.1, 0.5 and 1.0 steps.

Manishgupta200, 2022-05-07:
Please aassign it to me.. . I'm able to solve it i think