fossasia/pslab-desktop #533: Home Screen: Show Instrument Tiles in responsive way depending on screen size

mariobehling, 2019-08-29:
Currently the instruments on the home screen are not responsive and show up in two columns. Please change this in a way that depending on the screen size or window width one, two, three or more columns could be displayed. When implementing a responsive way ensure the UI does not get too small to view for the user.
[image]

imskr, 2019-09-01:
I would like to work on this!  Please assign.

AakashMallik, 2019-09-01:
> I would like to work on this! Please assign.

@imskr Go ahead

pushpak1300, 2019-09-03:
Can I work On it?

imskr, 2019-09-03:
@pushpak1300 I'm on it!

asha15, 2019-09-17:
@imskr Are you working on this?

darkthunder007, 2019-09-21:
if anyones not working on this , can i?

daya2940, 2019-09-21:
can i work on this

darkthunder007, 2019-09-21:
Oh, I also want to work on it...

CloudyPadmal, 2019-09-23:
@asha15 @darkthunder007 @daya2940 one of you can take this up :+1:

asha15, 2019-09-24:
Shall i work then?

CloudyPadmal, 2019-09-30:
@asha15 yes you can :+1:

asha15, 2019-10-05:
@CloudyPadmal  sir i cannot find the code for the homepage. I'll appreciate of you could help me

adityastic, 2019-10-05:
@asha15 https://github.com/fossasia/pslab-desktop/blob/development/src/App.js

asha15, 2019-10-05:
Thank you @adityastic

arghac14, 2019-10-29:
@mariobehling @AakashMallik @CloudyPadmal is this issue still active, can I work on this?

adityastic, 2019-10-29:
@arghac14 yes it is

arghac14, 2019-10-29:
@adityastic can i work on this?

adityastic, 2019-10-29:
@arghac14 sure, go ahead!

ayushiitr2901, 2019-12-03:
Is this issue still open, I would like to work on it.

NimishGopal, 2020-01-18:
I wanna work on this.

NitinMadhukar, 2020-01-21:
@adityastic can i work on this

shivanandaalva, 2020-05-20:
Hey,
I would like to work on this

PR #565 (2020-06-13): create reponsive multicard layout home screen
* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [ ] Bug fix
- [x] Feature implementation
- [ ] Doc updates

* **What changes have you introduced?**
I have shown Instrument Tiles in responsive way depending on the screen size
Fixes  #533

* **Does this PR introduce a break [... cut]
closed 2020-06-13 by orangecms