fossasia/pslab-desktop #535: UI fragments/artifacts/pixelation on Home Screen

mariobehling, 2019-08-29:
The lines connected with the icons in the home screen show fragments or artifacts/pixelation. This is probably due to the graphic formats and limited resolution. For the web it is now possible to use SVG vector graphics, which are small and scalable without loosing any quality. Please research and implement icons as SVGs or find another way to avoid artifacts in the UI.

[image]

srivastava9, 2019-09-01:
Hey Man I am newcomer here.Would like to work on this issue please assign.

AakashMallik, 2019-09-06:
> Hey Man I am newcomer here.Would like to work on this issue please assign.

Go ahead

AakashMallik, 2019-09-06:
@srivastava9

priyanshunayan, 2019-11-06:
@adityastic @mariobehling can I work on this issue?

adityastic, 2019-11-06:
@priyanshunayan go ahead

PR #557 (2019-11-28): Remove Pixelation
* **Please check if the PR fulfills these requirements**
- [x] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [x] Bug fix
- [ ] Feature implementation
- [ ] Doc updates

* **What changes have you introduced?**
Fixes #535 
I have removed the pixelation issue that was initially occurring. 

* **Does this PR introduce a breaking change?**
No.

* **Preview / Steps [... cut]
closed 2019-11-28 by mariobehling