fossasia/pslab-desktop #556: Missing module canberra-gtk-module

cweitat, 2019-10-01:
* **I'm submitting a ...**
  - [ x] bug report
  - [ ] feature request

* **What is the current status?**
Missing canberra-gtk-module when running a new install from the snap store for Ubuntu 18.04LTS

* **If the current behavior is a bug, please provide the steps to reproduce it**
Install PSLab from Snap Store
Start the application through terminal by typing "pslab"
Error message of "Gtk-Message: Failed to load module "canberra-gtk-module" appeared twice
Application opens up but unable to detect PSLab

* **What is the expected behavior?**
Everything to work

* **Please tell us about your environment:**
 Ubuntu 18.04.2LTS

* **Other information** ( Any research that you may have done which you think is the cause of problem )
Ran several commands to reinstall / install the module but still facing the same error

* **Would you like to work on it?**
Open to anyone

cweitat, 2019-10-01:
https://github.com/lcherone/LXD-UI/issues/5#issuecomment-396045533
Might be an issue due to Electron

CloudyPadmal, 2019-10-01:
Yea I've seen this warning with few other desktop apps. I'm not sure if we can fix it from our side.

orangecms, 2020-06-14:
The  warning can be safely ignored. If the app cannot connect to the board, it is likely because of the corresponding Python library or the udev rules. I am not sure how the Snap distribution channel was set up. We are currently overhauling the build process and reconsidering distribution channels. The codebase is now at a state where you can build and run the app again. Please see #607 for detailed instructions. Thank you!

closed 2020-06-19 by cweitat