fossasia/pslab-desktop #589: Revert: update deployment on dropbox

mariobehling, 2020-01-15:
The update to store installation files on Dropbox has not been discussed. Files that are below 100 MB should be generated automatically and stored in the install branch.

The autogeneration of install files does not work anymore in the install branch. This should be fixed first. The access to the Dropbox account needs to be owned by the organization.

Therefore please revert: https://github.com/fossasia/pslab-desktop/pull/585

adityastic, 2020-01-15:
@mariobehling at the moment the file size of only linux installation is under 100 mbs. The package size will increase over time and the linux deb is on margin by 1.8mb. The folder can be shared with any dropbox account, hence I can give access to your account easily.

mariobehling, 2020-01-18:
@adityastic As long as the size is below 100 MB we are fine. Please make a PR where we store the deb file in the install branch. Furthermore please create a Readme with the links to the Dropbox files. There is no need for a snap file currently as it is not working.

PR #590 (2020-01-18): chore: revert dropbox deployment
Fixes #589
closed 2020-01-18 by mariobehling