fossasia/pslab-desktop #594: dmg file for Mac is not available in installation branch

kumuditha-udayanga, 2020-01-20:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**
dmg file for Mac os is not available in installation branch

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs

* **What is the expected behavior?**
dmg file should be on installation branch

* **Please tell us about your environment:**
  - OS with version number: [ Windows / OSX / Linux Distro ]
N/A

* **Other information** ( Any research that you may have done which you think is the cause of problem )

* **Would you like to work on it?**
Yes

kumuditha-udayanga, 2020-01-20:
I'm working on it

adityastic, 2020-01-20:
@kumuditha-udayanga the file is larger than 100mbs that's the reason it isn't there, it will be deployed to a dropbox account. I'm changing the deployment in the upcoming days

closed 2020-01-20 by adityastic