fossasia/pslab-desktop #605: Project not running locally

Dishebh, 2020-01-30:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**
Currently, the console throws an error:

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs
[image]

* **What is the expected behavior?**
The console should be from any errors / warnings. 

* **Please tell us about your environment:**
  - OS with version number: [ Windows / OSX / Linux Distro ]

* **Other information** ( Any research that you may have done which you think is the cause of problem )

* **Would you like to work on it?**
Yes

MuhammadWasif, 2020-03-24:
The problem is not permanent. Probably, there is an issue in installation of the application. Please consider cloning and installing again. Please Try again. There's no problem with code base.

PS: I also had the same issue but it got solved when I re-installed the app.

CloudyPadmal, 2020-04-01:
Thank you @MuhammadWasif for pointing out a solution.

closed 2020-04-01 by CloudyPadmal