fossasia/pslab-desktop #606: Downloads are broken

AnielShri, 2020-04-02:
* **I'm submitting a ...**
  - [x] bug report
  - [x] feature request

* **What is the current status?**
Unable to download windows instaler

* **If the current behavior is a bug, please provide the steps to reproduce it**
The download links from [https://pslab.io/downloads/](https://pslab.io/downloads/) point to the "[instal](https://github.com/fossasia/pslab-desktop/tree/install)l" branch on this repository. 

* When downloading the repository as a zip from github, the windows binary shows up as a 1KB file.
* When cloning the git repository, the windows binary shows up as a 1KB file

* **What is the expected behavior?**
Make the binaries avaiable under "[releases](https://github.com/fossasia/pslab-desktop/releases)". Or on your pslab.io server. It feels dumb that I bought your hardware and am unable to use it because you can't provide working software for it.

* **Please tell us about your environment:**
  - Windows

* **Would you like to work on it?**
No

orangecms, 2020-12-05:
We now have builds for all platforms using GitHub Actions (see #652). Please verifiy the pre-release and see if it works and report your platform etc here!

See https://github.com/fossasia/pslab-desktop/releases/tag/v2.5.0

cynthi8, 2020-12-06:
The pslab_2.5.0_amd64.deb package works (board connected and basic multimeter measurements) on my Kubuntu 20.04

orangecms, 2020-12-07:
The download has been verified to work on Windows 10. Should you run into trouble, please open a new/specific issue. Closing this now. :)

closed 2020-12-07 by orangecms

urbatecte, 2020-12-07:
You rock it !
Thanks à lot !