fossasia/pslab-desktop #607: Failure while building the project.

Hyperion101010, 2020-04-18:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**
Getting compilation errors while building the project. "window.require" is not a function.
I installed the npm libs through a normal npm install in the terminal.

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs
Just do normal installation of the project and hit npm start.

* **What is the expected behavior?**
The pslab shall start on localhost:3000

* **Please tell us about your environment:**
  - OS with version number: [ Windows / OSX / Linux Distro ]
Ubuntu 19.10 

* **Other information** ( Any research that you may have done which you think is the cause of problem )
I have made some changes and tried compiling it. 
First when the error is there the screenshot (below)
[image]

* **Would you like to work on it?**
Yes

EkanshMangal, 2020-05-07:
Hello Sir, I would like to work on it.
If issue is still open Can i try it?

adityastic, 2020-05-07:
> Hello Sir, I would like to work on it.
> If issue is still open Can i try it?

Go ahead 🙂

orangecms, 2020-06-14:
Hey there, thanks for the report!

I've looked into the build process and reworked quite some bits. Please update to the latest development branch and run the following:

```sh
npm install
npm run react-build
npm run pack
```
That should create a directory named `dist/` with the app in it.
Run `./dist/linux-unpacked/pslab` to launch it.

For interfacing with the device, make sure to have the `pslab-python` library installed as well as the udev rules. For Arch Linux, I have added packages to the AUR. Wth `yay`, a simple `yay -S python-pslab` suffices. (Remark: by convention, python comes first for the package naming.)

Please let me know if everything works for you. If you have a device and everything is set, you should be able to use the wave generators, for example. An LED and a resistor can be used to check. Thanks a lot! :)

orangecms, 2020-06-14:
Another note: For development, please run two sessions in your shell:

- `npm run react-start` for the React development server
- `npm run electron-start` for the Electron frontend application

This has hot reload included, so when you edit files, they are automatically updated in the Electron app window. Good luck! :)

orangecms, 2020-06-14:
Should you get an error about the sandbox, please try the following:
`sudo sysctl kernel.unprivileged_userns_clone=1`

Some distributions have not yet enabled the necessary kernel options otherwise. See also https://github.com/electron/electron/issues/17972 for details.

orangecms, 2021-02-25:
With the newly introduced cross env support, which makes the app run on CMD and Powershell, and due to inactivity, I am closing this now. Thanks!

closed 2021-02-25 by orangecms