fossasia/pslab-desktop #610: Receiving malformed JSON on device reset

orangecms, 2020-06-13:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**

When I click on the reset button in the app, it does say "device reset" and "device connected" in a snackbar at the bottom. In the terminal emulator from which I am running the app, I can see some errors.

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs

```
20:02:57.457 › SyntaxError: Unexpected token / in JSON at position 0
    at JSON.parse (<anonymous>)
    at PythonShell.<anonymous> (file:///home/dan/Projects/fossasia/pslab-desktop/background_tasks/linker.html:28:30)
    at PythonShell.emit (events.js:182:13)
    at /home/dan/Projects/fossasia/pslab-desktop/node_modules/python-shell/index.js:324:22
    at Array.forEach (<anonymous>)
    at PythonShell.receiveInternal (/home/dan/Projects/fossasia/pslab-desktop/node_modules/python-shell/index.js:322:15)
    at PythonShell.receive (/home/dan/Projects/fossasia/pslab-desktop/node_modules/python-shell/index.js:296:21)
    at Socket.emit (events.js:182:13)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:260:13)
20:02:57.458 › SyntaxError: Unexpected token R in JSON at position 0
    at JSON.parse (<anonymous>)
    at PythonShell.<anonymous> (file:///home/dan/Projects/fossasia/pslab-desktop/background_tasks/linker.html:28:30)
    at PythonShell.emit (events.js:182:13)
    at /home/dan/Projects/fossasia/pslab-desktop/node_modules/python-shell/index.js:324:22
    at Array.forEach (<anonymous>)
    at PythonShell.receiveInternal (/home/dan/Projects/fossasia/pslab-desktop/node_modules/python-shell/index.js:322:15)
    at PythonShell.receive (/home/dan/Projects/fossasia/pslab-desktop/node_modules/python-shell/index.js:296:21)
    at Socket.emit (events.js:182:13)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:260:13)
```

* **What is the expected behavior?**

no error :)

* **Please tell us about your environment:**
  - OS with version number: [ Windows / OSX / Linux Distro ]

Arch Linux

* **Other information** ( Any research that you may have done which you think is the cause of problem )

I am using the PSlab Python library v1.0.1.

* **Would you like to work on it?**

cynthi8, 2021-01-23:
The malformed string is `'Radio transceiver not installed/not found'`. 

The PSL library prints it when the device is connected and the NRF24L01 module is initialized. [With this recent refactoring](https://github.com/fossasia/pslab-python/commit/bb3a5e7f2349ea01589b54f5f4c44910cf843a24#diff-061c5cf67820c3b946e14a1f221a033e1cbcc37ca617fd0d13219c9072fcee5dL977), it now uses logging and that should solve the issue because INFO level logs do not go to stdout (by default). I tested this change, and the bug was gone.

The next pslab python update should fix this bug.

cynthi8, 2021-02-05:
This malformed JSON is no longer received.

orangecms, 2021-02-05:
Awesome, thanks for checking! So I'm closing this  :)

closed 2021-02-05 by orangecms