fossasia/pslab-desktop #615: Fix the Data Logger screen layout

orangecms, 2020-06-16:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**

On smaller screens, the text is cut off although there would be enough space to expand the cards.

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs

[image]

* **What is the expected behavior?**

The text should be fully readable.

19-pragya, 2020-08-15:
can i work on this issue

orangecms, 2020-08-18:
Yes, just go ahead and file a PR. :)

Defining fixtures and setting up / using Storybook would help a lot, you can look into that also.

Manishgupta200, 2022-05-07:
I want to work on it. please assign it to me