fossasia/pslab-desktop #618: read data from sensors via I2C

orangecms, 2020-06-20:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current status?**

With #612, I2C can now be scanned for sensors. The next step is to implement reading from them.

* **Other information** ( Any research that you may have done which you think is the cause of problem )

Check what is implemented in PSL (https://github.com/fossasia/pslab-python) and how it works. It might need some work as well.

* **Would you like to work on it?**

19-pragya, 2020-08-15:
can i work on this if it is available??

orangecms, 2020-08-15:
Sure, just file a PR! I have drafted the basics, will create and link a draft PR.

orangecms, 2020-08-15:
see #623 for basic functionality

orangecms, 2020-08-16:
Note that only a handful of sensors are actually implemented in the Python library. I have started to work on support for a Time of Flight sensor in https://github.com/fossasia/pslab-python/pull/130.