fossasia/pslab-desktop #621: Add documentation for Mac

Yukti-09, 2020-07-22:
* **I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request

* **What is the current status?**

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs

* **What is the expected behavior?**

* **Please tell us about your environment:**
  - OS with version number: [ Windows / OSX / Linux Distro ]

* **Other information** ( Any research that you may have done which you think is the cause of problem )

* **Would you like to work on it?**

raj5036, 2020-08-19:
Can I work on this?

orangecms, 2020-08-28:
Hi there, we will redo quite some bits around the docs; we already have some more up to date instructions at
https://docs.pslab.io/desktop/Readme.html. Please have a look there for now.

If you are running macOS and would like to help out, you are most welcome. :)
Please also share how you prefer to install the necessry dependencies since macOS is lacking a native system package manager for software distribution; Homebrew, MacPorts, extracting tarballs, .dmg images, installers, extra shell scripts (such as nvm for Node.js) are all possible; however, they are user-specific preferences and we should keep it their choice thusly. A little guidance for people with less experience leading to a somewhat maintanable procedure would be good, I think. Thank you!

orangecms, 2020-08-28:
No specific assignment is necessary - just file PRs please so we can discuss them @raj5036 @Yukti-09 :)

orangecms, 2020-12-05:
FYI: We now have builds for all platforms using GitHub Actions (see #652).

I edited the issue to be about the development setup.