fossasia/pslab-desktop #629: display tutorials in a view

orangecms, 2020-10-10:
Currently, the app doesn't tell much itself how to use the board etc. We have some views showing the front and back.

Idea: Add an entry to the menu to shows the tutorials from docs.pslab.io :)

ritik307, 2021-01-03:
@orangecms  hey, i would like to work on this issue can you please elaborate about it a little. showing example image on how you want it to look like would be a great help

orangecms, 2021-01-03:
Sure :) A simple way to go about this is to render an iframe and embed the docs page in it. There is a menu (the burger menu on the top left) where you can add the navigation entry. The other option is to open a new window, like the "Buy PLSab" entry does.

ritik307, 2021-01-04:
@orangecms thanks for the explaination. I will start working on it . Can you please assign it to me.

ritik307, 2021-01-04:
**Sir while running the application using `npm start` I get the following error can you please help me out in figuring what the issue really is. I have tried uninstalling and installing node again but it still didn't work**

[0] ''npm' is not recognized as an internal or external command,
[0] operable program or batch file.
[1] 'run' is not recognized as an internal or external command,
[1] operable program or batch file.
[2] 'react-start'' is not recognized as an internal or external command,
[2] operable program or batch file.
[3] ''wait-on' is not recognized as an internal or external command,
[3] operable program or batch file.
[4] 'http:' is not recognized as an internal or external command,
[4] operable program or batch file.
[2] react-start' exited with code 1
[1] run exited with code 1
[0] 'npm exited with code 1
[4] http://localhost:3000/ exited with code 1
[3] 'wait-on exited with code 1
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! pslab@2.5.2 start: `concurrently 'npm run react-start' 'wait-on http://localhost:3000/ && npm run electron-start'`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the pslab@2.5.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

orangecms, 2021-01-04:
What's your OS? How did you set up npm?

ritik307, 2021-01-04:
@orangecms on windows sir

orangecms, 2021-01-04:
Okay, @Abhijay007 is also having trouble with the setup on Windows.

I suggest running the two scripts that are run through `concurrently` separately, just manually. So, in one terminal, `npm run react-start`, and in another, `npm run electron-start`.

orangecms, 2021-01-04:
If the two of you can figure out a fix for the setup on Windows, that would also be great to share with others. Feel free to open issues here and post details on your setup; i.e., how exactly you installed Node and npm, what your `PATH` looks like, whether you use WSL or possibly WSL 2, how you set that up, and so on. It is a bit complex on Windows, unfortunately; sorry for the mess.

Abhijay007, 2021-01-04:
> If the two of you can figure out a fix for the setup on Windows, that would also be great to share with others. Feel free to open issues here and post details on your setup; i.e., how exactly you [...]

ok @orangecms sir, we will try to solve this error together

ritik307, 2021-01-04:
@orangecms sir I have tried running the application with `npm run react-start`  and `npm run electron-start` in different WSL terminals but it's still showing error

[image]

Abhijay007, 2021-01-04:
> @orangecms sir I have tried running the application with `npm run react-start` and `npm run electron-start` in different WSL terminals but it's still showing error
> 
> [image]

@ritik307, I am getting this same error

orangecms, 2021-01-04:
I have created https://github.com/fossasia/pslab-desktop/issues/663 so that you can keep the discussion there, since this issue here is not related and about the tutorials view.