fossasia/pslab-desktop #640: Increase frequency limit for square waveform generator

bessman, 2020-10-26:
The PSLab is capable of outputting PWM signals at frequencies up to 8 MHz. However, the GUI currently does not go above 5 kHz. The frequency slider in the waveform generator view should go to 8 MHz for square waves. It should still be limited to 5 kHz for analog waves.

partha2000, 2020-10-28:
@bessman Could you assign this issue to me? I can solve this issue. Thanks in advance.

cynthi8, 2021-04-06:
@bessman could you point to some documentation where 8MHz is the limit for the square wave generator?

From the code it looks like 10 MHz is the limit.

https://github.com/fossasia/pslab-python/blob/bbd18d18911e7bb6fa8430cfe0b287cd9517fa88/pslab/instrument/waveform_generator.py#L342

bessman, 2021-04-07:
It's not really a hard limit at a specific frequency. The timing accuracy of the pulses starts to degrade above 1 MHz. So either 8 MHz or 10 MHz is fine as the software limit.

PR #681 (2021-04-10): fix (waveform): Increase square frequency limit to 10MHz
* **Please check if the PR fulfills these requirements**
- [X] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [X] Bug fix
- [ ] Feature implementation
- [ ] Doc updates

* **What changes have you introduced?**
Increased square frequency limit to 10MHz

Fixes #640 

* **Does this PR introduce a breaking change?**
No

* **P [... cut]
closed 2021-04-10 by orangecms