fossasia/pslab-desktop #646: migrate away from remote module

orangecms, 2020-11-01:
See https://www.electronjs.org/blog/electron-10-0#breaking-changes

`linker.html` reports the warning:

```
Electron Deprecation Warning The 'remote' module is deprecated and will be disabled by default in a future version of Electron. To ensure a smooth upgrade and silence this warning, specify {enableRemoteModule: true} in the WebPreferences for this window.
```

This needs to be taken care of for #624 also.

cynthi8, 2020-12-19:
Also in #613. Is this a priority for us? We don't run any external code so the security isn't a huge concern.

orangecms, 2020-12-19:
It has been deprecated and will be removed either way. We are already at Electron 11 now. Removal is planned for 14: https://github.com/electron/electron/issues/21408

The choice is not only about security, but also about maintenance. Whether it is a priority to anyone depends on your personal preference for outdated software. After all, for example, noone could any longer use the former Qt application, because Qt 4 is no longer being distributed. My personal preference is to keep applications maintained and up to date. I wouldn't want to use outdated software.

cynthi8, 2020-12-19:
Ah ok, sounds good! I was looking into this, but was getting annoyed as many of the Electron docs still reference `remote` , like [here](https://www.electronjs.org/docs/api/dialog).

closed 2021-04-10 by orangecms