fossasia/pslab-desktop #652: Setup Github Actions and Substitute Travis CI

mariobehling, 2020-12-05:
Travis CI stopped working. Please substitute Travis with Github Actions.

orangecms, 2020-12-05:
Done, please checks if the builds work for you: See https://github.com/fossasia/pslab-desktop/releases/tag/v2.5.0

orangecms, 2020-12-05:
I have verified the 2.5.0 tar.gz release to work. \o/

cynthi8, 2020-12-06:
pslab_2.5.0_amd64.deb works (board connects and basic multimeter tests) on Kubuntu 20.04

orangecms, 2020-12-07:
Picking up from Gitter: You may need to create a link because PSLab Desktop tries to call `python3` specifically.

- Windows: `mklink python3.exe python.exe` (in the Python install directory)
- UNIX-like systems: `ln -s python python3`

orangecms, 2021-02-25:
We are at 2.6.0 now and builds are working just fine. Closing this; possible issues can be reported separately.

closed 2021-02-25 by orangecms