fossasia/pslab-desktop #653: Provide a Chocolatey package

orangecms, 2020-12-05:
This is a convenient way for people to install the software on Windows.

See https://docs.chocolatey.org/en-us/create/create-packages

AnshMishra2001, 2020-12-15:
@mariobehling I am getting this error while downloading Chocolatey, even after disabling real-time protection.
At line:1 char:1
+ [System.Net.ServicePointManager]::SecurityProtocol = 3072; iex ((New- ...
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
This script contains malicious content and has been blocked by your antivirus software.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : ScriptContainedMaliciousContent

orangecms, 2020-12-16:
That is very unfortunate, I'm sorry. False positives may happen. You can compare checksums and verify signatures to check for integrity.

AnshMishra2001, 2020-12-17:
I have mangaged to download Chocolatey on windows and now working on the editing the files. In the file chocolatey.install file 
$fileLocation = Join-Path $toolsDir 'NAME_OF_EMBEDDED_INSTALLER_FILE'. What should I replace with NAME_OF_EMBEDDED_INSTALLER_FILE

orangecms, 2020-12-17:
Could you elaborate? I only find a reference to that at https://docs.chocolatey.org/en-us/guides/create/create-custom-package-templates. Is it necessary to create a custom template? I would expect the default ones to suffice. What _exactly_ have you tried, what did you read in the documentation, what worked and what didn't, etc?

AnshMishra2001, 2020-12-17:
PS C:\Users\EDMUND SIMON\OneDrive\Desktop\MyTest\package> **choco install package -dv -s .**
Chocolatey v0.10.15
Chocolatey is running on Windows v 10.0.18363.0
Attempting to delete file "C:/ProgramData/chocolatey/choco.exe.old".
Attempting to delete file "C:\ProgramData\chocolatey\choco.exe.old".
Command line: "C:\ProgramData\chocolatey\choco.exe" install package -dv -s .
Received arguments: install package -dv -s .
RemovePendingPackagesTask is now ready and waiting for PreRunMessage.
Sending message 'PreRunMessage' out if there are subscribers...
[Pending] Removing all pending packages that should not be considered installed...
Performing validation checks.
Global Configuration Validation Checks:
 - Package Exit Code / Exit On Reboot = Checked
System State Validation Checks:
 Reboot Requirement Checks:
 - Pending Computer Rename = Checked
 - Pending Component Based Servicing = Checked
 - Pending Windows Auto Update = Checked
 - Pending File Rename Operations = Checked
 - Pending Windows Package Installer = Checked
 - Pending Windows Package Installer SysWow64 = Checked
The source '.' evaluated to a 'normal' source type

NOTE: Hiding sensitive configuration data! Please double and triple
 check to be sure no sensitive data is shown, especially if copying
 output to a gist for review.
Configuration: CommandName='install'|
CacheLocation='C:\Users\EDMUND SIMON\AppData\Local\Temp\chocolatey'|
ContainsLegacyPackageInstalls='True'|
CommandExecutionTimeoutSeconds='2700'|WebRequestTimeoutSeconds='30'|
Sources='.'|SourceType='normal'|Debug='True'|Verbose='True'|
Trace='False'|Force='False'|Noop='False'|HelpRequested='False'|
UnsuccessfulParsing='False'|RegularOutput='True'|QuietOutput='False'|
PromptForConfirmation='True'|AcceptLicense='False'|
AllowUnofficialBuild='False'|Input='package'|AllVersions='False'|
SkipPackageInstallProvider='False'|PackageNames='package'|
Prerelease='False'|ForceX86='False'|OverrideArguments='False'|
NotSilent='False'|ApplyPackageParametersToDependencies='False'|
ApplyInstallArgumentsToDependencies='False'|IgnoreDependencies='False'|
AllowMultipleVersions='False'|AllowDowngrade='False'|
ForceDependencies='False'|Information.PlatformType='Windows'|
Information.PlatformVersion='10.0.18363.0'|
Information.PlatformName='Windows 10'|
Information.ChocolateyVersion='0.10.15.0'|
Information.ChocolateyProductVersion='0.10.15'|
Information.FullName='choco, Version=0.10.15.0, Culture=neutral, PublicKeyToken=79d02ea9cad655eb'|

Information.Is64BitOperatingSystem='True'|
Information.Is64BitProcess='True'|Information.IsInteractive='True'|
Information.UserName='EDMUND SIMON'|
Information.UserDomainName='LAPTOP-7PBHSR7N'|
Information.IsUserAdministrator='True'|
Information.IsUserSystemAccount='False'|
Information.IsUserRemoteDesktop='False'|
Information.IsUserRemote='True'|
Information.IsProcessElevated='True'|
Information.IsLicensedVersion='False'|Information.LicenseType='Foss'|
Information.CurrentDirectory='C:\Users\EDMUND SIMON\OneDrive\Desktop\MyTest\package'|

Features.AutoUninstaller='True'|Features.ChecksumFiles='True'|
Features.AllowEmptyChecksums='False'|
Features.AllowEmptyChecksumsSecure='True'|
Features.FailOnAutoUninstaller='False'|
Features.FailOnStandardError='False'|Features.UsePowerShellHost='True'|
Features.LogEnvironmentValues='False'|Features.LogWithoutColor='False'|
Features.VirusCheck='False'|
Features.FailOnInvalidOrMissingLicense='False'|
Features.IgnoreInvalidOptionsSwitches='True'|
Features.UsePackageExitCodes='True'|
Features.UseEnhancedExitCodes='False'|
[... 36 lines cut ...]
By installing you accept licenses for the packages.
Using '.'.
- Supports prereleases? 'True'.
- Is ServiceBased? 'False'.
package not installed. The package was not found with the source(s) listed.
 Source(s): '.'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more
 assistance.

Chocolatey installed 0/1 packages. 1 packages failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Failures
 - package - package not installed. The package was not found with the source(s) listed.
 Source(s): '.'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more
 assistance.
Sending message 'PostRunMessage' out if there are subscribers...
Exiting with 1

AnshMishra2001, 2020-12-17:
This is the error I am getting after having edited the files

AnshMishra2001, 2020-12-17:
I was able to create the Chocolatey package. Now I have to add the package folder to the code and generate a PR?

AnshMishra2001, 2020-12-17:
I have created the package

orangecms, 2020-12-17:
Nice! Did you publish it to the community repo?

I.e., will it appear on https://chocolatey.org/packages?q=pslab?

inomag, 2021-01-06:
Hi there, is the issue still open ?

cynthi8, 2021-02-24:
@inomag I believe the issue is closed. @AakashMallik could confirm?
I am able to install v 2.5.2 from [https://chocolatey.org/packages/pslab-desktop](https://chocolatey.org/packages/pslab-desktop) and v 2.6.0 is awaiting moderation currently.

orangecms, 2021-02-25:
Yup, 2.5.2 was approved about 10 days ago. Does someone know how Chocolatey gets notified about updates? Evebtually, the outcome here should be a few lines in the README about the install process for ens users and (if applicable) the release process for maintainers. Can we note something down? :)

orangecms, 2021-04-29:
As an update: The package is available at https://community.chocolatey.org/packages/pslab-desktop and maintained at https://github.com/tunisiano187/Chocolatey-packages.

orangecms, 2021-04-29:
I've dropped that on the README as well. Closing this now. :)

closed 2021-04-29 by orangecms