fossasia/pslab-desktop #655: Python 3 may not be found though installed

orangecms, 2020-12-07:
On some systems, by now, `python` refers to Python 3, while on others, it may refer to Python 2. However, PSLab Desktop only tries to call `python3`.

We need to check the defaults across various systems. Maybe another workaround is necessary. We are using the `python-shell` module. There could be a nice solution by now taking care of pointing to the right version of Python.

cynthi8, 2020-12-08:
python-shell calls python3 by default [link].(https://github.com/extrabacon/python-shell/blob/f3b64d3307d8dc15eb9c071d8aa774c1e7d5b2d7/index.ts#L94)

orangecms, 2020-12-08:
Nice, thank you for digging it out! :)

Then we can just remove the explicit option to use `python3`.

It's in https://github.com/fossasia/pslab-desktop/blob/development/background_tasks/linker.html#L10 and the other background task. Could you file a PR? Thanks!

cynthi8, 2020-12-08:
> Could you file a PR?

Doing it right now!

orangecms, 2020-12-08:
Can you also confirm that `py` refers to Python 3 on Windows?

https://github.com/extrabacon/python-shell/commit/9baa598beb0a5c8409bf61907a5ec3bc3d176162 looks like that's what they have been calling it for at least two years now. :)

cynthi8, 2020-12-08:
Yes, I just tested that a few minutes ago ;).
[image]

orangecms, 2020-12-08:
Thank you a ton!

closed 2020-12-08 by orangecms