fossasia/pslab-desktop #658: Oscilloscope stops working when disabling X/Y Plot

alcorhythm, 2020-12-16:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**
Oscilloscope does not work anymore after disabling X/Y Plot.

* **If the current behavior is a bug, please provide the steps to reproduce it**
1. Go to Oscilloscope
2. Activate channels 1 and 2, observe working
3. Enable X/Y Plot
4. Disable X/Y Plot
5. Observe no longer showing any signals

* **What is the expected behavior?**
Oscilloscope should keep working after deactivating X/Y plot.

* **Please tell us about your environment:**
Windows 10

* **Other information** ( Any research that you may have done which you think is the cause of problem )
Steps to make the Oscilloscope work again: disconnect and reconnect USB.
It does not help to reset the device, exit and re-enter the Oscilloscope view or restart the desktop application.

cynthi8, 2020-12-17:
Reproduced bug on Kubuntu 20.04. I will look into this.

cynthi8, 2020-12-17:
Looking at the console logs which are shown with `npm start`, it doesn't seem like the Disable X/Y Plot is working properly. Even after the toggle switch is deactivated, X/Y Plot data is collected instead of the typical Oscilloscope data.

[log_during_normal_oscilloscope.txt](https://github.com/fossasia/pslab-desktop/files/5706940/log_during_normal_oscilloscope.txt)
[log_during_xy_oscilloscope.txt](https://github.com/fossasia/pslab-desktop/files/5706943/log_during_xy_oscilloscope.txt)
[log_after_xy_oscilloscope_is_deactivated.txt](https://github.com/fossasia/pslab-desktop/files/5706944/log_after_xy_oscilloscope_is_deactivated.txt)

PR #660 (2021-01-17): Fix #658 - Oscillosope not disabling X/Y Plot properly
This fixes the bug where the oscilloscope would not return to normal
operational mode after the X/Y plot was disabled.

* **Please check if the PR fulfills these requirements**
- [X] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [X] Bug fix
- [ ] Feature implementation
- [ ] Doc updates

* **What changes have you introduce [... cut]
commit 2bbb9eaa (2021-01-17): Fix #658 - Oscillosope not disabling X/Y Plot properly

This fixes the bug where the oscilloscope would not return to normal
operational mode after the X/Y plot was disabled.
closed 2021-01-17 by bessman