fossasia/pslab-desktop #665: Logic Analyzer display, functional and unit issues

alcorhythm, 2021-01-14:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current status?**

Using the Logic Analyzer to grab a 1MHz SPI signal, using all four channels (capturing the SDI, SDO, SCK and CS of a SPI flash memory chip in this case).

 - Most of the time when hitting Analyze, it will only show some small condensed data on the right side of the screen, with most of the screen left blank.
 - The triggering settings don't seem to have any effect.
 - The X axis of the graph should be using microsecond units (μsec), but it says mSec (which is milliseconds and is inconsistent with how it is spelled for Capture Time, ms).
 - Clicking on the Capture Time to enter it using the keyboard brings up a popup, but you cannot enter any value into it.
 - The Capture Time slider allows 0.01 to 1 ms, the popup asks for 0 to 5.
 - What does the Capture Time do anyway?

* **If the current behavior is a bug, please provide the steps to reproduce it**
> A screenshot of the issue may be sufficient for UI bugs

Most of the time when hitting Analyze, we get nothing, simply because the Analysis stops before any events occurred and there is no triggering.

If we do get something, then most of the time, this is what we get:
[image]

Only by hitting the Analyze button many times, you may get lucky and see something more useful:
[image]

If you are not in control of the device that you are measuring, this is pretty much unusable. In this case I am specifically sending the same thing over and over on the SPI, with other devices on the bus physically disconnected, in hopes of capturing something.

* **What is the expected behavior?**
 - Scaling so that the data is displayed in a useful way
 - Correct and consistent units and values
 - Some meaning/functionality to the triggering settings
 - Explanation of, and correctly functioning, Capture Time setting

It would also be nice if you can name the channels ;)

* **Please tell us about your environment:**
  - OS with version number: [ Windows / OSX / Linux Distro ]
Windows 10

* **Other information** ( Any research that you may have done which you think is the cause of problem )

In the firmware there seems to be a limit on the maximum amount of time between events, and the total number of events that can be captured. When using 4 (or 3) channels, the time limit is (much) shorter than for 2 (or 1) channels. This time limit only depends on the number of channels captured. That only deepens the confusion about the meaning of Capture Time.

It would be better if we can just keep capturing until some trigger event actually happens. That is, if no events are detected, just capture again, until something is captured.

* **Would you like to work on it?**
No... (well I would, but I have no time any time soon).

bessman, 2021-02-05:
I will address those issues that pertain to pslab-python.

> * Most of the time when hitting Analyze, it will only show some small condensed data on the right side of the screen, with most of the screen left blank.

This is due to https://github.com/fossasia/pslab-firmware/issues/56. Essentially, the logic analyzer sometimes/often returns false timestamps at the start of a capture series. pslab-python tries to remove them, but cannot always distinguish false timestamps from real ones. When that distinction cannot be made, pslab-python defaults to returning possibly false timestamps, rather than removing real ones.

> * The triggering settings don't seem to have any effect.

Last I checked this was working. I'll check again this weekend.

> * What does the Capture Time do anyway?

Exactly what you might expect: The logic analyzer runs for <capture time> before returning any captured timestamps.

bessman, 2021-02-05:
OK, I took a look at the code, and the behavior you're seeing with regards to the long initial period of no logic level changes is actually the intended behavior. It happens when the trigger condition to start the capture is disabled, i.e. when capture begins immediately. In this situation, the initial state (high/low) of the digital input is recorded as timestamp 0.

All that means it is possible to fix this in pslab-desktop by just not plotting the first data point. I'll patch it.

orangecms, 2021-02-20:
Duh, it said "partly fixes", and GitHub nicely interprets that and closes the issue when merging a PR. Sorry!

PR #670 (2021-02-20): Logic analyzer: Don't plot first data point
This prevents a long period of nothing interesting happening from being plotted, which used to take up valuable screen real estate. Partly fixes #665.
PR #677 (2021-04-10): Feat (Logic Analyzer): Auto Recapture and Capture Time slider
Capture time slider is changed to allow for larger values and more
fine-tuned control. Auto Recapture checkbox added to have the logic
analyzer continuously recapture until an event is detected.

The Graph component is receiving the LA data and if a capture has no
event within it and the auto recapture is on, the LA is toggled
back on.

Standarized UI to use μs which is converted to ms for the backend communication.

Fix (Logic Analyzer): Define inputCheck

The CustomDialog dispatch needs an inputCheck function to be defined.
The Lo [... cut]
closed 2021-04-10 by orangecms