fossasia/pslab-desktop #684: App doesn't connect to device on Mac

cynthi8, 2021-04-20:
* **I'm submitting a ...**
  - [X] bug report

* **What is the current status?**

Opening the app via normal means will not connect to the device.

* **If the current behavior is a bug, please provide the steps to reproduce it**

From Gitter - @chiefenne
> I installed the desktop app which runs fine. I also installed the Python library. When I connect PSLab via a USB cable and then start the desktop app, it says "no device connected". I validated the correct installation of the Python library with the script given in the docs (psl.multimeter.measure_capacitance() ...). This worked fine. But the device does still not show up in the desktop app.

Current workaround - @chrvoigt
> start it from the command line $ open /Applications/PSLab-Desktop.app .. just make sure to connect the device first 

* **What is the expected behavior?**
Connection works without needing this workaround.

* **Please tell us about your environment:**
  - OS with version number: [ Windows / OSX / Linux Distro ]
Mac

* **Other information** ( Any research that you may have done which you think is the cause of problem )

* **Would you like to work on it?**
I don't have a Mac so it will be hard for me to reproduce.

chiefenne, 2021-04-20:
Here is some information on my environment where I encounter above described behavior.

Environment:
macOS Big Sur: 11.2.3 (20D91)

Python:
I am using Python 3.7.8 from an Anaconda installation.

For which environmental variables do I have to look out?

cynthi8, 2021-04-20:
> For which environmental variables do I have to look out?

PSLab desktop opens a Python Shell to run the backend that comminicates with the PSLab board. The Python Shell runs whatever `python3` binary is found on the `PATH`.

Running GUI applications from the command line will pass the correct `PATH` that contains the python3 executable. Running via Finder/double clicking the icon in /Applications will not pass this variable.

chiefenne, 2021-04-20:
Ok. Thanks

chrvoigt, 2021-04-21:
just adding my 2 cents ... the notion of 'standard python' in macOS may be confusing 

macOS comes with a baked-in Python2.7 https://docs.python.org/3/using/mac.html 
$ which python
/usr/bin/python
 
if we install Python3 with brew it has a standard path for installing python versions 
$ which python3
/Library/Frameworks/Python.framework/Versions/3.9/bin/python3

if we install PSLab with pip it also has a standard path for installing PSLab under 
/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages

so maybe that could help to improve findability of Python and / or PSLab libraries.  

however is there maybe an easier way of bundling Electron and Python code? (even at the expense of the size of the packaged App?)