fossasia/pslab-desktop #690: Upgrade to Electron 13

orangecms, 2021-05-31:
We may need to upgrade gradually; currently, we are at version 10.

See https://www.electronjs.org/blog/electron-13-0

Neerajrawat123, 2024-01-06:
Assign this to me...

orangecms, 2024-01-06:
Just file PRs.

This is already two years old now. By now, I suggest moving away from running Python in a separate process and instead use the WebSerial API and run the code in a Wasm context. Ideally rewrite the PSLab Python library in a portable language such as Rust and let the Python library just call into it. Or look at Pyodide as an intermediate solution.

bessman, 2024-01-06:
FYI I've begun moving functionaly from pslab-python / pslab-android into the firmware. The serial protocol willl be reimplemented from scratch in protobuf as part of this.

Once the new protocol is stable, I plan to write a common backend for desktop and android in rust.

This will create a lot of code churn. Any front end work should wait until the new API is ready.