fossasia/pslab-desktop #697: Expand to a monorepo, libsigrok bindings

orangecms, 2022-01-29:
## Current State

Currently, the app is set up to depend on the [pslab-python library](https://github.com/fossasia/pslab-python). However, that is usually hard for end users to set up.

Now we have a WIP implementation to support [PSLab in libsigrok](https://github.com/fossasia/libsigrok), and besides using that in Sigrok's PulseView, we can also use it here in the PSLab Desktop App.

## Next Step

The first step is to create bindings for Node.js, then set up the IPC stack to forward data to and send requests from the Electron windows. With a monorepo (pulling in libsigrok as a git submodule), it becomes simpler to set up everything. Then we can gradually expand, and even provide a UI for other instruments etc supported by Sigrok.

orangecms, 2022-02-09:
Even nicer would be to wrap libsigrok in WebAssembly, so that we can reuse it for a second build that can be used on the web directly, say, `web.pslab.io`.

For a related example, see:
https://github.com/GoogleChromeLabs/web-gphoto2

orangecms, 2022-03-26:
I've forked @RReverse's libusb fork again: https://github.com/orangecms/libusb

Built it like this with Emscripten:

```
#!/bin/sh

./autogen.sh
 # emconfigure add things like --host=wasm32
emconfigure ./configure
emmake make
```
and got some output:

```
make  all-recursive
make[1]: Entering directory '/home/dama/Projects/FOSSASIA/PSLab/libusb'
Making all in libusb
make[2]: Entering directory '/home/dama/Projects/FOSSASIA/PSLab/libusb/libusb'
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_posix.lo
  CC       os/threads_posix.lo
  CC       os/linux_usbfs.lo
  CC       os/linux_udev.lo
  CXXLD    libusb-1.0.la
make[2]: Leaving directory '/home/dama/Projects/FOSSASIA/PSLab/libusb/libusb'
Making all in examples
make[2]: Entering directory '/home/dama/Projects/FOSSASIA/PSLab/libusb/examples'
  CC       dpfp.o
  CCLD     dpfp
  CC       dpfp_threaded-dpfp.o
  CCLD     dpfp_threaded
  CC       ezusb.o
  CC       fxload.o
  CCLD     fxload
  CC       hotplugtest.o
  CCLD     hotplugtest
  CC       listdevs.o
  CCLD     listdevs
  CC       sam3u_benchmark.o
  CCLD     sam3u_benchmark
  CC       testlibusb.o
  CCLD     testlibusb
  CC       xusb.o
  CCLD     xusb
make[2]: Leaving directory '/home/dama/Projects/FOSSASIA/PSLab/libusb/examples'
Making all in tests
make[2]: Entering directory '/home/dama/Projects/FOSSASIA/PSLab/libusb/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress
make[2]: Leaving directory '/home/dama/Projects/FOSSASIA/PSLab/libusb/tests'
make[2]: Entering directory '/home/dama/Projects/FOSSASIA/PSLab/libusb'
make[2]: Leaving directory '/home/dama/Projects/FOSSASIA/PSLab/libusb'
make[1]: Leaving directory '/home/dama/Projects/FOSSASIA/PSLab/libusb'
```

will see where that takes us :)

orangecms, 2022-03-27:
Note: There is also https://wicg.github.io/serial/ WebSerial, but it would be less portable to build on top of that.

normanr, 2022-04-09:
Note that browsers typically don't allow access to devices via multiple APIs, eg if a USB device supports HID then it's accessible via WebHID and not WebUSB (same for audio/storage/video/wifi etc). I expect that (probably eventually) WebUSB will not allow access for serial devices.

For serial devices (that use serial_open) you probably want to use WebSerial, for usb devices (that use libusb_open) you probably want to use WebUSB. For HID devices you'll probably have to use WebHID. SCPI device s you probably have to use WebSerial or WebUSB (depending on the transport). For TCP devices, you might be able to use Direct Sockets API (when it becomes available), otherwise Websockets with something like websockify.

orangecms, 2022-04-09:
Yea, good points! There's multiple ways for us to integrate; I've done some more research and put together a presentation I'm giving at FOSSASIA Summit in about an hour @normanr 

Slides are here: https://metaspora.org/pslab-goes-native-goes-web.pdf

Also, related PR regarding libusb: https://github.com/libusb/libusb/pull/1057

orangecms, 2022-05-14:
There’s a Pyodide JavaScript module:
https://www.npmjs.com/package/pyodide

It can simply spawn Python like we already do in the desktop app, but without the need for the extra module maintaining background tasks. :partying_face: