fossasia/pslab-desktop #708: error during app launch

Harshdev098, 2024-02-02:
* **I'm submitting a ...**
  - [ y] bug report
  - [ ] feature request

* **What is the current status?**

I have installed all the packages to run the app but its encountering an error 

[image]

I thought it could be a version issue, I have uninstall the node_modules folder and reinstalled but didn't get satisfactory result. Although  the desktop app has been installed 

[image]

Also, while running build script it fails to make build folder as mentioned in the repository readme file   
https://github.com/fossasia/pslab-desktop?tab=readme-ov-file#building-the-app. 

[image]

ralphtheninja, 2024-05-06:
Yeah I have the same error. Is this project no longer active?

bessman, 2024-05-06:
The PSLab project as a whole is active, but pslab-desktop is unmaintained.

ralphtheninja, 2024-05-06:
> The PSLab project as a whole is active, but pslab-desktop is unmaintained.

Roger that. Are there any types of software being developed that's supposed to replace this desktop software?

bessman, 2024-05-06:
We have a GSoC student working on improving the Android app. As for desktop, I have some ideas but am currently busy working on the firmware.

ralphtheninja, 2024-05-15:
@bessman I'd be happy to help out. This project is awesome!

bessman, 2024-05-17:
Great! If you can get the app back to a functional state, I'd be happy to review and merge a pull request. I'm not sure what exactly is broken right now, and I don't have much experience with the JS parts of the app (I mainly work with the PSLab's firmware (in C) and the Python layer).